victorlucachi/void_ergo #1: Keycaps vs Layout

leo-vwdilab, 2021-01-28:
Sorry to open an issue but seemed like the quickest way to get in contact 😉

Interested in building one **void ergo** of my own and was wondering what keycaps did you use (or which ones can someone use) that would not impact with the layout?

Btw: Kudos for the design ⌨️🙏😉

victorlucachi, 2021-01-28:
Hey, and thanks! No worries, it is quite an effective way to get in touch after all, haha.

The plates are compatible only with Cherry MX style switches, so any compatible keycaps (that use the appropriate stems) should do the trick; all of the keys used in the layout are 1u. I personally used cherry profile, dsa and xda, and I did not encounter any issues.

Good luck with your build, and if you need any help dont hesitate to get in touch.

leo-vwdilab, 2021-01-28:
Many thanks, will keep you posted 😉

closed 2021-01-28 by leo-vwdilab

fairglen, 2021-03-04:
Hi @victorlucachi 
Just dropping by to say thanks for sharing this project of yours 🥇 👍 
This was the [result ](https://github.com/fairglen/qmk_firmware/tree/master/keyboards/handwired/void_ergo)

Do you have a paypal account or something, so that I can donate at least enough for a coffee?
Keep on 🤘

victorlucachi, 2021-03-05:
> Hi @victorlucachi
> Just dropping by to say thanks for sharing this project of yours 🥇 👍
> This was the [result ](https://github.com/fairglen/qmk_firmware/tree/master/keyboards/handwired/void_ergo)
> 
> Do you have a paypal account or something, so that I can donate at least enough for a coffee?
> Keep on 🤘

Hello!

Thank you for printing it, I love the end result! The orange filament is so vibrant! Hope you find it comfortable enough for daily use. 
Personally I've been going back and forth between a Corne and my Ergo S, and I might be biased here, but I think I like the latter better.

Thanks again, and [a cup of coffee](https://paypal.me/victorlucachi) is much appreciated!