victorlucachi/void_ergo #2: What's length of M2 brass inserts do I need to buy?

Eloren1, 2021-07-31:
> 10 x M2 brass inserts (3.5mm outer diameter)

But what about length? 

[image]

victorlucachi, 2021-07-31:
Hey, the ones I've used were 3mm. The insert holes themselves are about 6.7mm in depth though, so you could probably use longer inserts, but the max that I'd go with would be around 5mm, because some molten plastic will build up in the bottom of each insert.

Eloren1, 2021-08-04:
And another question, what's this? 
[image]

Eloren1, 2021-08-04:
@victorlucachi can't reopen this issue btw

victorlucachi, 2021-08-04:
> And another question, what's this?
> [image]

You probably missed it in the readme, it's a mounting slot for a 6x6x6 push button, used to reset the master side.

Eloren1, 2021-08-04:
Also I cannot find this layout in QMK Configurator. Nothing by searching `handwired/void_ergo`

victorlucachi, 2021-08-04:
> Also I cannot find this layout in QMK Configurator. Nothing by searching `handwired/void_ergo`

My builds are not added to the QMK repo, so no QMK Configurator support. You'd have to use my fork of QMK, which is also linked in the Readme, but it's not up to date and there have been some breaking changes ever since I last updated it; you'd probably have to fix any issues that may arise on your own, for the time being.

closed 2021-08-04 by victorlucachi

Eloren1, 2021-08-04:
Maybe I can use other QMK layouts like for Atreus 50

Eloren1, 2021-08-05:
@victorlucachi May I fork this project and add it to QMK Configurator? Your project will be linked in the readme. Keyboard name: Elrgo S. 😄

victorlucachi, 2021-08-05:
Reasoning behind this?

The keyboard wont get added to the QMK Configurator in a day or two, it takes months for a PR to get merged. You can fork and edit the files for the firmware as you see fit though.

I'm currently reworking the models for the entire keyboard range, after which I plan on going through the firmware and creating a PR for all of them myself.

Eloren1, 2021-08-05:
I just wanted to add configurator support. I will wait for your PR. But why do you want to rework models?

victorlucachi, 2021-08-05:
I want to make some small improvements to them, such as:

- moving on to smaller hardware on the other keyboards besides the Ergo, so M2 screws and heat inserts (with the current design the cases would get stress fractures after a while, depending on the screws used)

- moving the controller to be centered on the case and maybe a modular controller slot

- a better, recessed design for the USB port

- and a bunch of other stuff

victorlucachi, 2021-08-05:
As a sidenode, when I get to the firmware I plan on adding support for [Vial ](https://github.com/vial-kb) too. It's similar to VIA Configurator, but a bit cooler in my opinion.