victorlucachi/void_ergo #3: button port question

UwUham, 2021-10-15:
heyo, i was just wondering about this part in particular, cause i feel like it won't print well without supports. Am I supposed to try to insert the button mid print or something to help it stay up?

[image]

Thanks in advance!

UwUham, 2021-10-16:
[image]
do you think that this would be an adequate support beam for the part? sorry, i'm kind of a novice at 3D printing

victorlucachi, 2021-10-16:
No worries. I'm using Prusaslicer, it's my favorite slicer, and I'd set up something like this:

[image]

a scaled down support enforcer, that would cover the entire shape of the slot. when sliced, it would look like this:

[image]

patrickmcquay, 2021-10-19:
I just printed a set of these, they print surprisingly well, considering the small features. I too use prusaslicer, and the default supports work great.

Be careful getting that support material out though.

closed 2023-06-13 by victorlucachi