victorlucachi/void_ergo #4: Wiring the audio connectors

108085, 2022-02-15:
Hi there,
I'm currently getting all the parts together and I can't wrap head around how to wire the audio connectors to the pro micros.
I also only got 5pin audio connectors, but that shouldn't be a problem.
Is it just VCC, GND, TX and RX to the pins or the audio connectos and the same on the "slave" side, or do i have to do more/something else?

I would really appreciate a little push in the right direction.

Sorry for the maybe dumb question, but my only experience with hand wiring is the Void9 ;)

108085, 2022-02-15:
Nevermind... After days of search, I think i found the solution, minutes after posting the "issue"
https://gist.github.com/mtei/b1515eba3d20086da262c83948f8e9a9#soft-serial-connection
In your firmware you're just using D1 instead of D0.

Correct me if I'm wrong.

I leave that here, in case someone else has the same question.

victorlucachi, 2022-02-15:
Hey! Yes, that's it.

You can wire the audio connectors pins in whatever order you like, same mapping for the left/right half; gnd, vcc and d1, by default.