victorlucachi/void_ergo #5: Question about the material list info

renatosuero, 2022-05-20:
Hi @victorlucachi ,
first of all, thanks for sharing the amazing keyboard. I'll build that(I've already have the case, if you want some real photos I can send you).
I have some questions about the inserts and screws, can you help me, please?

We have a brass inserts in the list, but it doesn't mention the length, what do you suggest for that?
10 x M2 brass inserts (aliexpress [link](https://www.aliexpress.com/item/4000585933306.html), 3.5mm outer diameter
[image]
)

another one I'd like to confirm is that one
10 x M2x12 allen head screws (aliexpress [link](https://www.aliexpress.com/item/32966941844.html), 14mm overall length, 4mm diameter head)
I didn't get if I need the 12 or 14mm
[image]

renatosuero, 2022-05-20:
thank you very much. I'll order the missind components them I'll come back with case and keyboarb 
btw the right side the bottom isn't equal the left one. I didn't realized when I've printed, for me it doesn't affect the usage so I'll go with that one
[image]

victorlucachi, 2022-05-20:
it's not equal in the sense that it's missing the controller mount and usb port? yes, that was intended, because I didnt want/need a port on my right hand. you can mirror the left half in your slicer of choice if you need easy access to the usb port though.

renatosuero, 2022-05-20:
yes it makes sense, there's no reason to reset or connect into the promicro. I'll build it soon and spam you with tons of photos  😂

renatosuero, 2023-04-14:
after long time, I'm back :joy: I'm finishing the keyboard, I'll be back with photos and try to help more in other issues

closed 2023-06-13 by victorlucachi