victorlucachi/void_ergo #7: Requesting additional Case & Switchplate model filetypes (namely .STEP for the dimensional data & encoding) for those of us interested in better tailoring the design to tastes.

PancakeGoon, 2022-08-14:
Hey there, @victorlucachi! MASSIVE fan of all of your boards - _especially_ the original  VOID Ergo & Ergo S!

&nbsp;&nbsp;&nbsp;&nbsp; First of all, I've sent you an e-mail correspondence through your site's contact form that goes into deeper detail for my plans, but figured I'd open this public official request to gauge other's possible interest in it as well.

But basically, I've wanted to build one of these for a while now & have also been learning Fusion360 for keyboard-related projects and designs. I'm also someone that always likes to customize or modify my builds beyond original intention, and this is even more relevant with the open-possibilities or a handwired build - so I wanted to try my hand at tweaking the design of the case halves & plates to alter it into a sort-of unofficial "Goon-efied" Ergo S fork as design-practice, whilst keeping the nod to the original and also releasing the files myself. Also gives the masses the same customization freedom!  

The only issue I've had so far _(which is quite early-on, I admit... 🙃)_, __is the fact that .STL files are a _MISERY_ to edit or modify—in any serious way—in a 3D-modelling/CAD context...__  
&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp; No in-coded Dimensions/tolerance data—this is especially a pain with Fusion 360, when it ACTUALLY wants to import external files like DXF's or STL's correctly... it's still an issue for _confirmed-working_ models/vectors to import with bugged-out scaling, and good luck trying to edit that further as STL—they can't encode real curves, it's just surface data, etc. - which is fine for 3DPrinting, which is also the intent here, I get that! And .DXF's are fine for the plates.

\- But including .STEP (or even better' **Fusion's .f3d project files**, if you used the program =D - although I totally understand if you'd like to keep that depth of your IP to yourself!) would make this board a fantastic resource as a split case-design reference, as well as enabling the ability for fans like myself to further tune & perfect the design to our personal tastes and physical differences.

Thanks, dude!

victorlucachi, 2022-08-17:
hey there PancakeGoon,

sorry for the late reply, to begin with; the email contact form, and the website for that matter, are in total shambles. opening this issue was probably the best course of action.

thank you for the kind words, much appreciated! I can totally relate to your situation, the driving force behind these projects, among other things, was to learn some cad design.

I would usually suggest starting from scratch, as in my opinion it's always best to start with a clean slate instead of tweaking or figuring out someone else's mess, but I dont mind posting the .step files, if you do choose to alter the plates or the case.

[I'll attach them here](https://github.com/victorlucachi/void_ergo/files/9358002/VOID.Ergo.S.-.step.zip) (bottom, top and plate for the left half, only bottom for the right half, because the other parts can be mirrored), and add them to the repository later.

thanks for getting in touch, and It would also be awesome if you'd post the remixed files somewhere, maybe for other people to use if they find them of interest.

closed 2022-08-17 by victorlucachi

PancakeGoon, 2022-08-18:
@victorlucachi 
Oh man, you are just the best for this! Thank you so much for the speedy response, too. 👍 
And lol, I'll keep that in-mind about your site and contact from here on out, and you can be __sure__ I'll be uploading everything I come up with dude! I still have a couple other project cases and stuff in-line, but I'll definitely be sticking my nose in & out of this in the mean-time. :D