victorlucachi/void_ergo #9: About wiring guide

jmvega-pki, 2023-03-27:
The little wiring guide ins awesome BUT, could please upload the pinout of the micros for this project? I'm a little lost.
Thanks!!

daithanhtran2912, 2023-03-31:
Hi Victor! I'm not sure it is ok if I should create a new issue to ask about my questions so I'm gonna post it here:
- For the TRRS jack `serial wiring`:  I'm curious about what method should I go with `(Bitbang/Half-duplex/Full-duplex)`? To my understanding from QMK docs and your firmware, I should go with `Bitbang` method, but how about the `Half-duplex` and `Full-duplex` methods?
- Is that possible if I want to use the `Full-duplex` method, and what should I do with the firmware? Do I have to modify it?

Thank you!!!

victorlucachi, 2023-03-31:
hey @daithanhtran2912 

if you're using pro micros or any atmega32u4 based controllers you're pretty much stuck with bitbanging. half/full duplex usart is compatible only with arm controllers.

all the methods that you mentioned, as well as instructions to enable each of them, are documented [here](https://github.com/qmk/qmk_firmware/blob/master/docs/serial_driver.md).

I've personally used only avr controllers with qmk split keyboards, I cant say if there is any real benefit in using one method over the others, but if you plan on building an Ergo with arm based controllers you will definitely have to configure the firmware from scratch.

daithanhtran2912, 2023-03-31:
Hi @victorlucachi, thank you for your fast response.
Currently, I'm using `Pro micro` for my Ergo. As you mentioned above, pretty sure that I only have two options `bitbanging` and `I2C wiring`.
Now I'm curious about the `I2C wiring`. Is that possible to use it for this project?

daithanhtran2912, 2023-04-01:
Hi @victorlucachi! 

Thank you for the information and sorry if I make everything complicated. I'm not a fast typer. My typing speed is around 80-100 wpm. And yes, the latency is what I'm wondering about. Does the latency of the bit bang serial is something noticeable?

I'm also open to experimenting with something new. So if i2c could make the latency better, I think I'm gonna go with it XD. 

Finally, thank you for creating and making this keyboard available for everyone, it is a piece of art!!!

(I want to share my W.I.P Ergo but don't know where to post it. I haven't attached the top and the bottom case together)
[image]

victorlucachi, 2023-04-01:
it's all good @daithanhtran2912 and I was just curious, haha. there's nothing wrong with wanting to use i2c, and to be honest I only went with serial because it was the default option and it seemed to work for everyone else.

I for one cant notice any input lag with bit bang serial, I've used it with the Ergo and with my Corne builds, but these days I've moved on to zmk and ble wireless controllers (and I still cant notice any lag whatsoever, even though on paper it should technically be slower? I can still type at an average of 100 wpm).

I did a quick search and discovered some rather old talk on the qmk discord server, mentioning tests and changes [for the Let's Split keyboard](https://github.com/qmk/qmk_firmware/pull/7732) and the conclusion was in fact that out of the box serial is faster than i2c. there's probably more info regarding this on the server as well, but I'm not smart enough to dive into the intricacies of qmk configuring though.

if you want to switch to i2c you will have to wire the d0/SCL pin on the controllers to the trrs jacks and add the pullup resistors, comment out or remove [the following line](https://github.com/victorlucachi/qmk_firmware/blob/dev_void/keyboards/handwired/void_ergo/info.json#L21) and add `#define USE_I2C` to the `config.h` file. [this should cover it](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_split_keyboard.md#i2c-wiring), in case that I missed anything else.

thank you for the kind words, I appreciate it!

that's a really clean build, you really went the extra mile, that paintjob is impeccable! if you didn't wire it up already I suggest using magnet (or enameled) wire, as it is thinner than normal wire, and it should help a bit in regards with the limited space inside the case. I'm looking forward to seeing your completed project and your impressions on it!

the reddit ergomechboards/handwiredkeyboards, or the multiple mechanical keyboard discord communities would probably be great places to share it with other people too!

daithanhtran2912, 2023-04-01:
Hi @victorlucachi, thank you for your detailed instructions! 

Yeah, I haven't wire it up and I planning on using enameled wire as well (I just want to go a little bit crazy over here XD).

I will share the project on reddit when I'm finished.

Thank you so much for helping me!

renatosuero, 2023-04-14:
yesterday I finished all soldeding, (except the encoder yet) today I want to test all switches, once everything is good I'll return with encoders.

closed 2023-06-13 by victorlucachi