victorlucachi/void_ergo #11: TRRS Pins Clarification

codercodingthecode, 2023-06-14:
Needing some clarification with TRRS wires.

The Void Ergo uses
[image]

The TRRS pinout uses a different approach to share power with the slave board.

[image]

I understand we are using SDA D2 pin and D20/D19 for Void Ergo
On the TRRS pinout is different using GND and VCC

Do we need 5 wires[D2, GND, VCC, D19,D20] or 3 wires?[D2, GND, VCC ] ?

Can someone clarify the proper connection with the TRRS wires?

victorlucachi, 2023-06-14:
hey @codercodingthecode!

as outlined [here](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_split_keyboard.md#serial-wiring), and on the diagrams that you posted above, the serial connection between the central and peripheral half uses only 3 pins: VCC, GND and either PD0/PD1/PD2/PD3.

in our case, for the void ergo, the predefined config uses the PD2 pin, so you would need to wire up the trrs jacks with VCC, GND, PD2. as for PD19 and PD20, they have nothing to do with serial, and they're used for the encoders.

hope this clears things out

codercodingthecode, 2023-06-16:
Thanks @victorlucachi .
One more question regarding pinouts.

Keyboard vs pro micro

Left side [master]
The software will be compiled separated for left and right with handedness by eeprom
https://docs.qmk.fm/#/feature_split_keyboard?id=handedness-by-eeprom

Does this wiring seem accurate?

[image]

victorlucachi, 2023-06-16:
the 3d models assume that left half is central, I never messed with eeprom handedness, by default my fork uses [split_usb_detect](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_split_keyboard.md#hardware-configuration-options). same firmware for both halves.

diode direction is col2row.

the rows are correct, as per your diagram, but the columns are not. you will want to wire up your right half like so:

[image]

codercodingthecode, 2023-06-16:
Thanks for the clarification.
Based on what I’ve seen around when researching hand wired keyboards. It seems to be a common question for starters.

This is my first keyboard build. I’ve should’ve started with a no split I guess 😄

Anyhow, I still can’t get the right hand split to work probably.
I’ve wired the right half as described above. And I still get the keys all messed up. The left side is good though.
I’ve checked for shorts on both ends and everything seems good. 🤷‍♂️

[image]

codercodingthecode, 2023-06-17:
Actually, I had unsoldered and soldered the same way back up lol
I fixed it and now it works.

Thank you for the help and for the project!

victorlucachi, 2023-06-17:
awesome, glad to hear that it's working!

I've been handwiring boards for a couple of years now, and I still make these kind of mistakes, I guess it comes with the territory.

anytime, thank you for printing and building it, hope you'll find it enjoyable to use!

closed 2023-06-17 by victorlucachi