victorlucachi/void_ergo #12: Switch Plate holes are a tad too big

Aggeloz, 2023-10-16:
Im using a Prusa Mk3s+ with the prusa slicer and i discovered that the switch holes are a tad too big and the switches move way too much and dont hold on too well. What could i do to fix this issue?

victorlucachi, 2023-10-17:
the models switch holes measure 14.2 mm (.2mm extra for tolerance, because not all printers are equally calibrated). if you have some calipers on hand you could measure the holes, determine their dimensional accuracy, and adjust the .step file accordingly. another option would be to hotglue the switches in place.

closed 2023-10-17 by Aggeloz

R3ISilva, 2024-04-17:
I also have find these to be true on the plate left stl file, just fliped the plate right and reprinted it for thight fit.