victorlucachi/void_ergo #15: Advice on making a modified version as a first split ergo keyboard

clorl, 2024-08-21:
Hello, this repo hasn't a lot of activity so I hope this message gets to someone.

I'm looking to upgrade my cheap 20$ keyboard to a split mech but I'm on a budget. I thought about hand wiring it myself and found your design.

However, it doesn't quite fit my needs, the modifications I'd need would be : 
- Low profile Choc switches
- 58 keys (similar to the Sofle)

I have experience in electronics/CAD/3D printing and programming. I already assembled an open source hardware project (a [PhobGCC controller](https://github.com/PhobGCC/PhobGCC-HW)). I have lots of tools and basic electronic parts at home, a 3D printer and a local maker's lab, bit no switches or keycaps on hand.

How hard can it be to modify your design to fit my needs given my level of experience? Do you have directions or general advice on how to do this?

Thanks for the advice

victorlucachi, 2024-08-21:
hey clorl,

you seem pretty skilled. considering your needs, if you're good with CAD it might make more sense to start from scratch, and take design cues from my models, if it's something that you find appealing. in my opinion adapting the existing files to accommodate a larger number of keys would only add more headaches than necessary, because it will require lots of adjustments.

[ergogen](https://github.com/ergogen/ergogen) could make things quite easy, regardless if handwiring or choosing to use a pcb, for the initial layout of the keys and even for designing a case. I highly suggest looking into it.

clorl, 2024-08-21:
Thanks for the advice, however the point I'm not sure about is on what to put the switches if there's no PCB? I guess I should print some kind of plate for the keys to go on in place of a PCB right?

I'll look into more documentation on the subject, thanks for the reply