victorlucachi/void_ergo #17: i think i did something wrong

andrwui, 2024-10-24:
I started assembling the keyboard today, I already assembled the right half, the left one is still not finished.

I to plugged the assembled half of the keyboard to the computer, and entered vial, to test everything was ok.

The keyboard worked, but it showed the left keyboard being pressed, which i thought it was normal because the firmware uses split detection, but also the keys were mirrored, so if i pressed the most upright left key, it will show the most upright right key.

I think i messed up soldering the diodes:

For the right half, the diodes are facing < and for the left half diodes are facing > (they are mirrored, facing to center)

Are the keys being mirrored because of this? Or is it that i haven't plugged both halves?

If I did mess up, is there a way through firmware I can tell qmk that diodes are the wrong way? I know I can just mirror the layout in vial, 
but I just wanted to know if there is a fix firmware-wise.

Also, desoldering is not an option i think.

Thanks for the awesome keyboard!!!

victorlucachi, 2024-10-24:
Pictures of the diodes/soldering would help paint a better picture, but as far as I can tell from what you're describing, it shouldn't be an issue (diodes can be installed either row2col or col2row, and you can set their direction in firmware, but they have to be soldered the same way across both halves).

Are you sure that your column wiring to the controller is not mirrored on the right half? I've seen that happen before.

andrwui, 2024-10-24:
Diodes for the right half are soldered this way:

[image]

And the diodes for the left half are soldered this way:

[image]

The columns are wired to the controller from left to right on both halves, I think I got that right.

Thank you for the answer!

victorlucachi, 2024-10-24:
Diodes seem fine, it doesn't matter which way their ends are pointing, electrically speaking the circuit is the same. But yeah, as long as your controller wiring checks out, everything should be good (you could take a look at [this](https://github.com/victorlucachi/void_ergo/issues/11#issuecomment-1594200109) wiring diagram for the column order and compare it with yours).

andrwui, 2024-10-25:
I was indeed mirroring the columns on the slave...

Now i'm having another problem...

I got them both to correctly work separately. But when i connect the trrs cable to connect them, the slave does not work, i see it's getting powered, but keystrokes do nothing.

The master works just fine, but the slave does not.

I thought it would be a problem with the trrs port wiring, but I don't think it's the case.

For context, here's pictures of the wiring:

Master:
[image]

Slave:
[image]

victorlucachi, 2024-10-26:
disabling the encoders in config and building the firmware from my fork, or just flashing the precompiled vial hex that's uploaded to this repo? please try the latter too, if you haven't already.

FedeArre, 2024-10-27:
> disabling the encoders in config and building the firmware from my fork, or just flashing the precompiled vial hex that's uploaded to this repo? please try the latter too, if you haven't already.

@andrwui forgot to post the fix, the issue was with the 3.5mm connection not fully inserting because of the 90° input. Trimming a bit of the protective rubber allowed it to fit perfectly, and the slave started working without issues."

victorlucachi, 2024-10-27:
that's great news, I'm glad it's working